pashutk/chocofi #1: top.stl switch plate holes too big

hmngwy, 2021-12-07:
Hello, I printed the case but found that the switch fit isn't snug, after some measuring I found that the holes are 15mm in width and height. 13.8mm w*h is the ideal hole size for choc switches, even the 14mm for MX switches are a bit loose. I would modify this and send a PR but I don't have the time right now to learn how too, so just letting you know for now. :(

[image]

Forgive the ruler work, the ruler tool in FreeCad doesn't snap to a grid.

pashutk, 2021-12-22:
@hmngwy Sorry for a late reply!

Initially I extruded 13.9x13.9 cuts but they was too tight to me. I guess that can be caused by bad z-offset on my printer, so I've exported version of the top with 13.9x13.9 cutouts, can you please test it?

There is an stl:
[Top2.stl.zip](https://github.com/pashutk/chocofi/files/7764676/Top2.stl.zip)

pashutk, 2022-01-10:
I've also added 13.9 version case stl files as v0 to [case](https://github.com/pashutk/chocofi/tree/main/case/) folder and moved prev version to v0.

AlexanderLindkjaer, 2022-01-22:
Hi @pashutk, printed the 13.9 version and they fit perfectly 🙂

pashutk, 2022-01-24:
@AlexanderLindkjaer Thank you for testing, that's great news!
Seems like this issue should be closed now.

closed 2022-01-24 by pashutk