pashutk/chocofi #4: Screw size on case, possible issue with tolerance

AlexanderLindkjaer, 2022-01-24:
I @pashutk , love the keyboard nice work :) 

What screw size is intended for the case? On most cases M2 is the standard. The screw holes are 2.4 mm, so way too big for M2 screws. So I'm thinking you are either using bigger screws or its a tolerance issues on your printer you had to adjust for. Like the plate holes V1 issue.

I modified the v1 case to fit m2 screws, on a printer with tight tolerance (1.9 mm screw holes). Let me know if you want a pull request.

pashutk, 2022-01-28:
@AlexanderLindkjaer oh huge thanks, it's definitely the same problem as in #1 
But it also could be a M2.5. Let me check, I'm sure that holes should be for M2, and if that's the case I will update stls.
I think there is no need for PR since its easier to fix this issue in source file and then regenerate stls.
Thanks again.

AlexanderLindkjaer, 2022-01-28:
@pashutk Great! Let me know if you need me to test the case on my printer 🙂 

If you or anybody reading wants to know the tolerance on their printer, a test like this can be printed
https://www.thingiverse.com/thing:2318105

pfn, 2022-02-09:
https://www.thingiverse.com/thing:5239708 -- I recently remixed the case to add tenting holes, and also had to fix the screw hole sizing: it has a 1.1mm radius, where it needs to be 1mm; M2 screws drop in. If your printer is that off (v0 case) your extrusion is probably too high and/or your x/y steps got badly misconfigured.

dcardamo, 2023-06-12:
I found a bottom case where the hole size was corrected to be m2 on thingiverse.   Sharing in case that helps anyone else:
https://www.thingiverse.com/thing:5239708

pfn, 2023-06-12:
See the comment immediately above yours  :D

azzamsa, 2023-06-12:
Anyone had the fixed screw location but without tenting feet holes?