pashutk/chocofi #6: firmware?

lilactown, 2022-03-17:
is there an OSS qmk/zmk config somewhere I could clone?

lilactown, 2022-03-18:
ah, I finally figured out that I can use the `crkbd` config for QMK and `corne` config for ZMK

pashutk, 2022-03-18:
> ah, I finally figured out that I can use the `crkbd` config for QMK and `corne` config for ZMK

@lilactown  Yeah corne firmware is the way to go. Schematic for chocofi is basically the same as fifi and fifi is using corne firmware too.

This is worth to be mentioned  somewhere in the Readme, so I'll keep this issue opened until then. Thanks!

zaddok, 2022-12-01:
So is the idea that we choose corne when we do the zmk `setup.sh` (which has 36 keys) but then inside the `corne.keymap` file, the left and right columns are ignored?

Thanks!

benjaminpotter, 2024-07-10:
Is there any plan to make resolve this issue? Providing this information in the README with complete instructions would be very helpful.