pashutk/chocofi #8: Are the provided gerbers original, or production gerbers

Cgboal, 2022-03-27:
I sent off the gerbers for this board to JLBPCB yesterday, and they got back to me asking the following: 

>Sorry to bother you, but there is an issue that we want to confirm with you before proceeding.
As shown below, It seems that the gerber file is working production file with compensation on the holes and trace for production.
If it is original file, we will make the compensation on the hole and trace. If it is production file, we will proceed without compensation.
For the file generated from the software, and has not been used for production, we call it original file.
So shall we proceed as per the gerber file with further compensation and optimization or not?

Are you able to shed some light on what the correct response is?

pashutk, 2022-03-27:
Hi @Cgboal!

Could you attach images they sent to you? Thanks

Cgboal, 2022-03-27:
Sure thing, thanks for the quick response
[image]

pashutk, 2022-03-27:
I wonder what compensations exactly they are talking about.

Anyway, this files were used to make a pcbs before, so I'd suggest you to proceed without any compensations. Please update the issue in case of any new information!

Cgboal, 2022-03-27:
Ok cool will do. Seems like from this doc on a different provider that different places handle compensation differently depending on their kit. https://www.pcbway.com/helpcenter/technical_support/Can_I_send_the_working_gerber_files_to_PCBWay_for_PCB_production_.html

Cgboal, 2022-03-28:
They also said that regarding the case designs, some places were very thin and risked cracking (red areas in pictures). Told them to go ahead with it anyway but I thought i'd post it here for posterity. 
[image]
[image]
[image]

pashutk, 2022-03-28:
Thank you for posting! That's unlikely will cause any problems especially when printing fdm or sla (not sure how mjf will perform though).

Cgboal, 2022-03-28:
Guess we’ll find out :D

pashutk, 2022-04-05:
@Cgboal damn that was fast.

Could you please share a photo of the black pcb, especially the region where controller should be?
And what about cases? Did they turn out well?

Cgboal, 2022-04-05:
Hey, just started putting this together. I bought through hole diodes and soldered them on, but seems like the switches then don't fit with the diodes in place. I'm guessing I should be using SMD's instead? Good thing I have a few spare boards

pashutk, 2022-04-06:
SMD is probably will have better fit for this case, yes. I think regular 1n4148-like diodes should work too since they are not as tall as hotswap sockets?

Btw I'm looking at the photo and it seems that they've made thicker traces. Could you please ask them if they did compensation on holes for the green ones?

Cgboal, 2022-04-06:
As far as I know they did compensation on the green one, but no compensation on the black ones. 

I've ordered some SMDs now which should work. Now I've got to find a battery that fits 🙄