pashutk/chocofi #13: Recommended battery version

zaddok, 2022-11-24:
What battery numbers do people find works best to slide under the nano? I’ve seen one person use the 301230.   Perhaps the ideal options could be listed in the readme for reference?

zaddok, 2022-11-24:
The nice nano documentation itself recommends the 301230, so thats probably the way to go for anyone looking:

https://nicekeyboards.com/docs/nice-nano/

pashutk, 2022-11-24:
Hi @zaddok!

Well, Chocofi may be used as a wired split without any need for batteries, so it depends on the controller you use.

But the general recommendation would be yes, use 301230 lipos, nice nano docs is a great source, and also there is Joric's wiki https://github.com/joric/nrfmicro/wiki/Batteries#301230.

zaddok, 2022-11-24:
Thanks @pashutk — I'm very excited to try my board (I have ordered a set from beekeeb so it all comes from in the mail at the same time,  I am just checking out what battery to buy).

pashutk, 2022-11-24:
I've added links to README (https://github.com/pashutk/chocofi/commit/49de38354cee7bea771488ee05ffbd5a3cc248c1), thank you for raising the issue.

That's amazing, hope you'll like it! Please contact me if there are any problems.