pashutk/chocofi #20: Question : Can the case be printed in metal?

nouritsu, 2024-01-23:
I want to build a keyboard that will last slightly higher temperatures, the place where I live comes slightly too close to the SLA max temperatures. Is the PCB good to go in a metal enclosure? If yes, are there any additional steps I have to take?

Zoltan-Balazs, 2024-09-22:
Did you manage to print it in metal?
I'm looking to do the same and wondering if anyone had success with it

leikoilja, 2025-02-05:
@nouritsu , may I ask, where do you live? 😅 

[image]

nouritsu, 2025-02-07:
I was under the impression that SLA printed articles need to be stored in an environment with temperatures below 30-35C. I used to live in a small coastal town in India, where temperatures would often go up to 43C during the harsher times in summer. I was afraid the case would deform if the temperatures got too harsh.

I understand that this is not the case, though 😅

I ended up building the [Temper](https://github.com/raeedcho/temper) keyboard and went caseless.

leikoilja, 2025-02-07:
haha, great choice on a temper tho! i have just finish my first build on the temper as well, lovely build! but i went with 3d printed casing.
@nouritsu , do you have pics of how your caseless looks? Did you plasticglass for the plates?

nouritsu, 2025-02-21:
Unfortunately I do not have the keyboard with me at the moment. However, here https://photos.app.goo.gl/T8FoAUFAczT2QjPv8 is an old photo.