pashutk/chocofi #23: Missing trace on the right side PCB (power switch-> RAW jumper).

articnet, 2024-11-14:
As per picture, if this trace is not connected, the right side does not currently switch on (can be solved by attaching the switch on the bottom and connecting the bottom jumper or by connecting a wire as in the picture). Am I missing something?

[image]

TannerGabriel, 2025-03-31:
Yes, I just had the same issue.  I could fix the board layout, but it looks like the schematic and board are out of sync. It seems like #9 only modified the board and didn't add the power switch to the schematic.

TannerGabriel, 2025-04-05:
@pashutk If possible, could you confirm that the schematic wasn't updated and if so do you still have the updated schematic file?